mxmxmx/O_C #5: Case corrections

conjure86, 2016-09-25:
Trying to compile via Linux, where it's case sensitive:

`/tmp/build472148504ceda71c5262b9ef872b0295.tmp/sketch/drivers/display.h:28:34: fatal error: SH1106_128x64_Driver.h: No such file or directory
 #include "SH1106_128x64_Driver.h"`

Calling to a case sensitive "SH1106_128x64_Driver.h" -- Fixed by renaming drivers/SH1106_128x64_driver.h to **SH1106_128x64_Driver.h**

conjure86, 2016-09-25:
Also, line 36 of OC_DAC.cpp calls:

` #include <spififo.h>`

should be recased to:

` #include <SPIFIFO.h>`

mxmxmx, 2016-09-25:
thanks.
will be [fixed](https://github.com/patrickdowling/O_C/commit/fa3d56d61601ce6846283a568dc2ca9e3b920067) in the next release.

mxmxmx, 2016-10-12:
should be fixed here now: 

https://github.com/patrickdowling/O_C/releases/tag/v1.1.2

closed 2016-10-12 by mxmxmx