mxmxmx/O_C #8: Gerber issue

conjure86, 2016-10-08:
In the latest Gerbers, **ornamentv3_2e_pastemaskbottom.gbp** is missing a pad. This will not affect PCBs being made, depending on the mfg, but will affect a stencil, etc.

See here: http://i.imgur.com/DzYMUAq.png -- Circled area.

mxmxmx, 2016-10-08:
ok, happens to be fixed in upcoming 2e* with enlarged pads for the encoders:

https://c2.staticflickr.com/6/5535/29888039100_55284c8aa4_z.jpg

closed 2016-10-08 by mxmxmx

conjure86, 2016-10-08:
Any change of it being fixed in the current v2e? I have a couple current PCBs and want to make a kapton stencil for them.

Thanks!

mxmxmx, 2016-10-08:
i'm not going to replace the gerbers with untested ones. the thing is made for hand-soldering, so stencils aren't really required unless you wanted to quasi-manufacture them, which we don't encourage. but if you want it, i can email you the 2e paste mask

conjure86, 2016-10-08:
If you could that would be great! conjure.music@gmail.com would be perfect. thanks!

I don't mind hand soldering, I just prefer to stencil if I can. Already have the tools to do so from years of other work. Thanks again!

mxmxmx, 2016-10-08:
ok, watch your spam folder