mxmxmx/O_C #18: Build guide gone?

markfrancombe, 2017-03-10:
Has the build guide at http://ornament-and-cri.me/ been removed for some reason? Was wondering about the 100ks and which they are.. Is it something to do with the arrows printed on the PCB?
Also construction tips for the teensie and oled... but can figure that out...

mxmxmx, 2017-03-10:
hi,
no, it's still there: hardware > build it!

http://ornament-and-cri.me/build-it/

closed 2017-03-10 by mxmxmx

markfrancombe, 2017-03-10:
OK thanks alot.
you might wanna change the link at the bottom of https://github.com/mxmxmx/O_C
It SAYS its to the build doc but aint! 

great project thx!

Mark

mxmxmx, 2017-03-10:
the build guide comprises more than just the step-by-step soldering stuff (BOM, troubleshooting, schematic, calibration, firmware update, and so on). so that's on purpose.