mxmxmx/O_C #56: Crash

Zek75, 2017-06-23:
Hi,
sometimes, after a while, when I'm in the quadrature mode, or dialectic ping pong, the module crashes. I mean the module doesn't response to any button, so I have to restart all my case. For now, I just use it home, to record, but it's difficult live. I use the 1.3.0 firmware. Do you know what I could do?
Thanks. And it's a really really great module!!!!!

mxmxmx, 2017-06-23:
hi
before we'll be trying to reproduce this -- did you use the .hex file from the releases page, or did you compile the firmware yourself? if so, which version of teensyduino did you use?

thanks

Zek75, 2017-06-23:
Thanks for your answer. In fact, I didn’t do it by myself, it was built by someone. I will ask him.

mxmxmx, 2017-06-23:
ok, i see. asking because first time i hear of the module crashing. (but there are potentially issues with using teensyduino 1.36). in the meantime i'd suggest you try with the latest code (1.3.1): 

https://github.com/mxmxmx/O_C

the update procedure (you want to use method "B") is described here: http://ornament-and-cri.me/firmware/

Zek75, 2017-06-23:
Ok thanks. So I can do the update by myself, it’s not too complicated?

Zek75, 2017-06-23:
And by the way, I have an other question. Is the Temps utile a good module to use with the O_C? To send trigs in it?

mxmxmx, 2017-06-23:
yep, that's the idea. it's not complicated. 

re: Temps utile. to use with OC it's as good as any clock module really. it depends on what you're after.

Zek75, 2017-06-23:
Ok thank you. I will try to do the update by myself…

themangoest, 2017-06-24:
Hi, i used the Teensy 3.2 with the o_C_v1_3_0.hex firmware. Haven't had the crash/freeze thing at the moment but will do some test running with the v1.3.1 here to see what happens.

mxmxmx, 2017-06-24:
so it crashed for both of you?

closed 2017-06-25 by mxmxmx