mxmxmx/O_C #57: Firmware update

Zek75, 2017-06-26:
Hi,
I tried to make the 1.3.1 firmware update. When I try method A, it keeps saying that the file is too large. With method B, it also says that o_c_rev.ino.hex is too large et the end, and the arduino software tells that "Teensy did not respond to a USB-based request to automatically reboot.
Please press the PROGRAM MODE BUTTON on your Teensy to upload your sketch."
Idf I press the button, nothing happens, the screen freezes. I swichted on the case as the website says. What can I do?

altitude909, 2017-06-26:
That's a hardware problem. Your teensy is not initializing and teensy loader is seeing it as a 2.0, not 3.2 (which has lees room).  Are you powering via USB or euro power?  If it was put together correctly, USB power should have been cut so its needs to be powered by your rack before plugging in the USB

Zek75, 2017-06-26:
Hi,
the module is plugged in the usb port, and the led is on. And I also power the euro. So you think the teensy is too old?

altitude909, 2017-06-26:
it has to be a 3.1 or 3.2.  Are you using the precompiled hex?  This sounds like a hardware problem, not a software one and should go to the forum, not the software bug report

Zek75, 2017-06-26:
Ok thanks

timchurches, 2017-06-26:
Please see point 8 of step 3 of method A on the firmware documentation page.

Zek75, 2017-06-26:
Hi,
I tried this, but it keeps on saying that the file is too large. What can I do?

mxmxmx, 2017-06-26:
"Teensy did not respond to a USB-based request to automatically reboot" would suggest you're using a cable meant for charging; i'd suggest try with a different cable.

Zek75, 2017-06-26:
Ha yes thanks!

closed 2017-06-26 by mxmxmx

Zek75, 2017-06-26:
Thanks a lot!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!