mxmxmx/O_C #63: piques euclidean cv control

evancorylevine, 2017-09-29:
The CV controllability of euclidean length, fill, and offset seem to be mixed up. When I set a CV input to control CV length, it would control fill. I also had an issue where it would seem to detect a voltage when there was nothing plugged in to control the CV "length," which was actually controlling the fill.

Just so you know, though, I'm deeply in love with this module. I built three and they are incredible. I will definitely be building more. Very excited about the terminal tedium and temps utile, as well.

patrickdowling, 2017-09-30:
At first glance it looks like the indexing in `EnvelopeGenerator::apply_cv_mapping` might be off-by-one since `CV_MAPPING_ADR` was inserted, leading to a mismatch between the `segments` array and the `CVMapping` enum. That'd explain some weird behaviour...

patrickdowling, 2017-09-30:
So assuming that theory is correct, this would fix it: https://github.com/patrickdowling/O_C/commit/f908b94d03d237273b8cdea1b56913367b22a4dd
That's on this branch: https://github.com/patrickdowling/O_C/tree/1.3.4dev
It seems to compile, but I can't test it currently :)

evancorylevine, 2017-10-01:
awesome! I won't be testing it tonight, but I found that the behavior was consistent across two of my units.

jddmagdaleno, 2018-01-07:
Strange, I've tried the proposed solution but unfortunately it actually only shifts the offset's direction, meaning Efill controls the Elenght, Eoffset controls Efill, etc.

patrickdowling, 2018-01-07:
Pfft, if I had a 1$ for every off-by-one, I'd probably have 0$ :stuck_out_tongue:
commit https://github.com/patrickdowling/O_C/commit/607734bded918d32cacb6a7173d62c9c98ecb01d on this branch
https://github.com/patrickdowling/O_C/commits/fix/envgen_mapping

jddmagdaleno, 2018-01-07:
the mapping now works correctly! Thanks for fixing it! :D

closed 2018-03-02 by mxmxmx