mxmxmx/O_C #68: LM1117-5 has no thermal pad

herrkami, 2018-01-21:
Hi, 
The LM1117-5 needs less thermal resistance on it's pad to stay within the recommended operating range. It might work like that but the component becomes annoyingly hot already and if the cabinet temperature exceed some 30 Celsius it goes out of range. You could improve that significantly by adding a thermal polygon around the top pad with thermal isolation disabled (see also p.18 in the datasheet of the LM1117). 
Cheers! Kami

mxmxmx, 2018-01-21:
Hi
thanks, bit late though. no further hardware revisions are planned ... if you're worried, maybe consider uOC. that said, if your LM1117 gets overly hot, chances are there's some other issue with your built.

closed 2018-01-21 by mxmxmx

herrkami, 2018-01-23:
OK, thanks!