mxmxmx/O_C #69: Invert screen option

analogboybcn, 2018-01-26:
I have 2 micro O_C.
One is in the upper side of my case, and the other is in the down side of my case.
I have had some ergonomics problem with the later, because all the cable end up above the knobs, making it difficult to operate.

It would be great to have an option to invert the screen images, so we can use the O_C flipped in a case.

Could this be possible?

mxmxmx, 2018-01-26:
see [here](https://github.com/mxmxmx/O_C/blob/master/software/o_c_REV/OC_options.h).

try uncommenting #define FLIP_180 and see if that works for you.

closed 2018-01-26 by mxmxmx

analogboybcn, 2018-01-26:
Hey! Thanks!

I’m not a dev at all, and compiling stuff is really not my strong point…
Is there a users place/forum where I can find a download to the latest firmware compiled with the screen flipped?

Thanks a lot.