mxmxmx/O_C #70: Running different apps at the same time?

simonvanderveldt, 2018-02-17:
Hi mxmxmx,

First of all: Thanks for the awesome modules you're creating and sharing with the world!

Wanted to ask if you think it would be possible/feasible to run multiple apps at the same time on O_C?

timchurches, 2018-02-17:
Simon,

Multiple apps at once would require a re-write of the underlying firmware app framework plus a re-write of every app to allow each and every input and output to be mappable, since all the inputs and outputs would need to be shared between two apps at once. That’s a very large amount of work which is unlikely to happen, I’m afraid.

Tim C

closed 2018-02-17 by timchurches

simonvanderveldt, 2018-02-17:
Hi Tim,

Thanks for the quick and clear reply!
Pity, but totally understandable :)