mxmxmx/O_C #74: Update to Teensy 3.6?

GitBruno, 2018-03-13:
Does it make sense to update this module to use Teensy 3.6 instead so we can load and save apps from the SD card?

patrickdowling, 2018-03-13:
Yes. No. Maybe. It depends :)
It's been discussed a few times, and it seems to be possible to hack a 3.6 on to the existing PCB.

Just throwing in a faster MCU and more memory doesn't immediately or automatically provide any massive benefits IMO though. But addressing the underlying limitations -- either hardware, like the shared SPI for DAC/OLED, or firmware, like the single-app paradigm and the lack of UI scalability --
 and the "update" quickly becomes a new module 😛 "load and save apps" smells like a re-write anyway...

GitBruno, 2018-03-14:
True that! 

The reason I am asking is that it feels that users are less engaging with creating new software for this module now the memory is full.

So it might be cool to create a app-store-esque interface on the website where you can select which apps you want to install and then download the hex/source/build-script

patrickdowling, 2018-03-14:
>The reason I am asking is that it feels that users are less engaging with creating new software for this module now the memory is full.

I doubt the full memory is causal. In the almost two years since release of this extended firmware, there hasn't (AFAIK) ever been broader interest in creating apps (or even implementing, not just suggesting, features) outside the main developers, with a few exceptions. Part of that is definitely chicken-eggy (the framework as more of an artefact is not designed to be particularly scalable, nor is it documented, which could be remedied) but it's a very small target audience in the first place. I'd also expect that with some effort in refactoring, one might even be able to free up some of the flash.

>So it might be cool to create a app-store-esque interface on the website where you can select which apps you want to install and then download the hex/source/build-script

Go for it. It's easy enough to do a selective-build mechanism (and it's come up on MW before as a workaround for dwindling flash as well as the lack of non-volatile storage), especially by converting to makefiles and ditching the Arduino IDE crutch -- which I still regret not doing, but at least it removes some support issues when setting up toolchains. The perhaps more challenging thing technically, as well as in terms of on-going human engagement, becomes how to collect/maintain/curate the apps in the "store." Do they all live in the main code repository? Or is there some way of linking to external repos, etc., which might require restructuring the codebase?

timchurches, 2018-03-14:
If there were many forked versions of the firmware each implementing additionally alternative  apps, then an effort to provide an “App Store” mechanism to consolidate them would be worthwhile, but as far as we know, there are no alternative versions featuring different apps.

GitBruno, 2018-03-19:
I guess having multiple forked firmwares is not the most elegant, but I understand you don't want to approve them all here :) Importing externals makes the most sense to me. This could be as simple as clone into the apps folder? Swapping out apps is not hard to do manually and I like the idea of a makefile.

I have not made any apps yet so maybe I'll start by creating a sample/boilerplate app we can clone? That way I can get a bit familiar and have a more useful chat later.

patrickdowling, 2018-03-19:
It might not work out of the box, but yeah, the one route could be to have a an `./apps/` folder where the apps exist as cloned or submodule folders, and are globbed by the build rules (assuming some better build system than Arduino, although maybe `./src/apps/` would work if they haven't changed to rules again). That's pretty rough around the edges though, to get there probably means cleaning up the app/framework dependencies, and it may be super-awkward for development and maintenance especially if (or rather, when) any app requires changes to shared code. And it requires some planning on how to (semi-)automatically register apps within the framework, as well as being able to selectively build a subset (which is probably easily done with the default being "all").

GitBruno, 2018-03-19:
Yeah, had a bit of a poke around and got a strong whiff of the rebuild smell. I am moving to the temps, loads of fresh space there :)

GitBruno, 2018-03-19:
I forgot to say, thanks! You guys are awesome

closed 2018-03-19 by GitBruno