mxmxmx/O_C #75: sequins channel B constantly runs, even without trigger inputs.

mtallen, 2018-03-22:
I've tested this on 2 O+C's, with the same behavior. 

repro steps:

1. Add clock input to TR1
2. Create a sequence for output A
3. Create 4 sequences for output B (may work with less have checked, this was my setup last night)
4. Set playmode for output B to SEQ+1 - SEQ+3
5. Set output B to respond to Clock TR1
6. Start clock into the OC
7. Stop clock 

When I do this output A will stop but output B continues on indefinitely. If you need more testing on this let me know.

mxmxmx, 2018-03-23:
Hi

I suspect you have set the channel B multiplier (mult/div) to some value > 1 ?

mxmxmx, 2018-03-23:
i see. that's an artefact of the multiplier stuff (= internal timer). it can be hacked away though (see corresponding [temps utile code](https://github.com/mxmxmx/temps_utile-/commit/e5840b97256541a7a7c26b3f95531f589d8c094b). i'll see to moving that over for the next firmware release

mtallen, 2018-03-23:
Thanks! Ill take a look at the temps code. Ive been meaning to try and set up an environment to get the OC stuff to compile locally. This is as good excuse as any.

nagruv, 2021-10-30:
Hello friends,
I am currently experiencing the same problem on the (OCP) VOR version. 
Do you know how to solve the problem ?

jmsiener, 2021-10-30:
@nagruv you should probably contact Plum Audio directly.  @Shayshez