mxmxmx/O_C #78: Piqued euclidian mode feature request

shurikenyo, 2018-09-04:
Can you add a reset all euclidian sequences to one of the CV in's?

shurikenyo, 2018-09-06:
sorry but i just read the 1.2 manual...from the 1.3 updates.

_Piqued
added a settable Euclidean trigger filter reset input, with user-specified clock division, on a per-channel basis. This permits lots of variation to be introduced in Euclidean rhythms, all driven from a single clock input._

Does this mean that i could use a reset gate from my midi to CV interface to reset the euclidian filter? Or any other gate for that matter? Or will it only use the a division from the original clock signal?

patrickdowling, 2018-09-06:
If I understand it correctly, for each channel you can use `Eucl reset` to assign one of the TRx inputs to act as a reset for the Euclidean pattern. That reset input has it's own clock division (`Eucl reset div`). So in theory you can assign all channels the same TRx and reset them simultaneously, or at different divisions of the same signal...

timchurches, 2018-09-06:
> If I understand it correctly, for each channel you can use Eucl reset to
> assign one of the TRx inputs to act as a reset for the Euclidean pattern.
> That reset input has it's own clock division (Eucl [...]

That’s correct. The idea was to add additional and different  pattern
variation to each channel, using the same clock/gate used to drive the
envelopes by specifying a clock division that is greater than the pattern
length. or separate clock inputs if you like.

shurikenyo, 2018-09-06:
Sounds perfect!! Nice to experiment with the division feature. Most of the time i think i will use only 1 reset for all euclidian filters and maybe use the other trigger inputs for different division clocks form my rotating clock divider is possible. but if not only 1 clock source for all sequencers is good also.

thanks

closed 2018-09-06 by shurikenyo