mxmxmx/O_C #79: Harrington 1200 root note gets stuck

altitude909, 2018-09-05:
Someone posted a video of this behavior: https://www.facebook.com/groups/euroracksynthesizers/permalink/2370070159677306/?comment_id=2370162729668049&reply_comment_id=2370167699667552&notif_id=1536152258755671&notif_t=group_comment

but I confirmed it on mine, (1.3.4), when selecting the Root (maj/min), changing the mode does NOT change the mode on top of the screen.  The work around was to change it, hold the encoder down untill it goes to mode select then go back to 1200 and it will change on the top row

mxmxmx, 2018-09-07:
... always a good idea posting on videos on facebook.

but thanks for the heads-up, we'll look into it.

mxmxmx, 2018-09-07:
... ok, so looks as if the mode is updated only if/when triggered (via TR1). is that the problem?

(quite a few modes still work this way, can be a bit confusing i know).

timchurches, 2018-09-07:
It may be more a lack-of-documentation issue - as Max says, most of the settings specify how or what transformations will occur in response to trigger inputs, but that isn’t explicitly stated in the documentation.

timchurches, 2018-09-07:
I have updated the manual to now read:

> The musicological mode of the root chord is set by Root mode in the menu to either major or minor. Chord inversion is similarly set using the Inversion menu item. Note that these settings are not immediate - they will take effect when the next transformation trigger input is received. The top line of the display indicates the current triad, not the current menu settings for these parameters. 

I think that’s correct (I am in Fukuoka, Japan, right now, no way of checking).

patrickdowling, 2018-09-08:
Yeah, I also think that's right, it's mostly editing the basis for the next transform, not the current state. Looks like the slight exception is if the CV sampling is set to `Cont`, in which case some changes (although not the transforms) are more immediate.

patrickdowling, 2018-09-08:
I'm not sure there's a "right" way to do it. The mode setting is the mode of the untransformed root triad, not what's on the outputs (which will depend on the chain of transforms so far) so it's pretty decoupled. A consequential approach could be to (also) reset the triad on UI change.

timchurches, 2018-09-08:
I have updated the documentation to reflect Patrick’s take on it - which I think is the logical choice and this it should be left as is.

mxmxmx, 2018-09-08:
Mmh, evidently it's confusing to people ... (and who is going to read the manual). But I'm not keen to "fix" it either, so fine with me ...

mxmxmx, 2018-09-08:
ok, so let's close this issue. 

@altitude909 : fwiw, here's a hack/variation, which should have the min/maj display change instantly both in the header and at the cursor (for the sake of UX "consistency", let's say), and resets the triad at the next trigger (whichever), just so that something happens, if something happens. i'm not sure about the rationale for this ... but then i don't know a thing about "Tonnetze".

https://github.com/mxmxmx/O_C/tree/135_dev

@timchurches  ... pleeeaaase, take it easy.

closed 2018-09-08 by mxmxmx

timchurches, 2018-09-08:
That’s just polite conversation in Oz...