mxmxmx/O_C #87: Possible to cap the cv at 5V?

Bahmsen, 2019-07-16:
Hello is it possible to use 1V/O but a Output Range from 0 to 5V?

mxmxmx, 2019-07-18:
in theory yes, sure. (if that's re firmware). in terms of the quantizer(s), the easiest would be to just restrict the input to 0V/5V

Bahmsen, 2019-07-18:
Thanks, all inputs would be in this range because the whole system works with 0-5V.

mxmxmx, 2019-07-18:
ah, i see. well in that case you're looking at both, hardware and firmware. the former is easy, you'd just have to scale the i/o suitably. the firmware / apps assume there's a 10 octave range though, so you'd have to adjust that, as well, which is also not overly difficult but tedious. (calibration menu, the individual apps ... )

closed 2019-07-18 by mxmxmx