mxmxmx/O_C #88: Right encoder

Kooza25, 2019-07-30:
My right encoder only become reverse when I reboot the module.
If in calibration I correct the direction it works good untile reboot.
Already tried to reflash the teensy

patrickdowling, 2019-07-31:
Did you save the calibration? That'd be the simplest explanation.
If yes, are the other values restored correctly?

Kooza25, 2019-07-31:
Of course it’s saved and out voltages remain calibrated

def1, 2019-07-31:
hey . i had the same on mine , tho cant really remember how it got fixed. i also recalibrated a few times after that it was gone i think.

patrickdowling, 2019-07-31:
Yeah, superficially it doesn't make sense since all the calibration data is saved in the same struct so it's a bit thin to base theories off. IIRC that part hasn't changed in years and if it were systematic it should be more apparent.

What hardware, software version? Does inverting the left encoder in calibration work after a reboot?

Kooza25, 2019-07-31:
Good input! After inverting left one too, it saved the right and so it does when putting them back to the right direction.
Problem solved, for now

closed 2019-07-31 by Kooza25