mxmxmx/O_C #92: Meta-Q puts out wrong scales / notes when changing scale slot

Triscus, 2019-10-02:
I like to use the scale slots in Meta-Q to use them as a kind of progression and I noticed some odd behaviour:

After a change of the scale slot occurs, the first note played sometimes doesn't belong to the old nor the new scale, sometimes the complete scale is wrong.

I tested it on two modules, tried the different ways to select the scales (trigger vs. cv addressing) and also used the latency settings all with the same result. 

- the transposition per CV aux is disabled

- Seq mode is TR+3 when used

- the trigger input is fed with 1/16 trigger at 110 BPM but it occurs also on slower triggers. 

- CV input comes from MI Tides (2017), the signal is attenuated and offset, sending the signal
through S&H doesn't make a difference.

- the issue happens in both channels

- the scale slots are set up as followed:
#1 A Pentatonic Minor (A,C,D,E,G)
#2 D Pentatonic Minor (D,F,G,A,C)
#3 F Pentatonic Major (F,G,A,C,D)
#4 E Pentatonic Minor (E,G,A,B,D)

- latest firmware is used

The glitch appears mostly while changing to scale #3 and to scale #4. I checked the output of the VCO with a fast tuner and it's clearly playing some sharp notes but there shouldn't be any in the scale. (And you clearly can hear it.) Thought about a calibration / voltage issue, but the sharp notes are perfectly in tune and the both modules are in different racks.

Any support on this is highly appreciated.

Cheers Triscus

mxmxmx, 2019-10-02:
hi

before we proceed: can you please try this branch (which includes some fixes to Meta-Q)
https://github.com/mxmxmx/O_C/tree/oc_p

Triscus, 2019-10-02:
Thank you for the fast reply. 

I compiled it without problems on the Arduino IDE. Only had time for a quick test, but it seems to work much better now. I noticed the notes were a couple of octaves lower with supposingly the same input voltage. 

I'll and do a cross test with a second o_c and firmware 1.3.5. and check on detail.

mxmxmx, 2019-10-02:
ah, sorry didn't realize the branch comes with the "VOR" option enabled:

you can either comment it out (here)[https://github.com/mxmxmx/O_C/blob/oc_p/software/o_c_REV/OC_options.h#L11] or use this [branch](https://github.com/mxmxmx/O_C/tree/dev136), that should fix the octave offset thing

Triscus, 2019-10-06:
Thank you, the issue went away with the new frimware, I used dev136 for a couple of hours

mxmxmx, 2019-10-07:
cool, thanks for checking. probably should pull this over into the master branch...

closed 2019-10-07 by mxmxmx