mxmxmx/O_C #101: Autoune tune not behaving correctly

djrrunstein, 2020-04-19:
Hi, I just built a Plum Audio OCP kit recently and I'm getting some weird behaviour in references mode when using the autotune function - it basically doesn't work. As it steps through the voltages, it doesn't seem to change the frequency of the oscillator, and then the autotune fails. If I listen to the output of the osc while going through the process, it sounds like OCP pings a voltage to the oscillator but doesn't sustain it. I also have a normal O&C and am doing exactly the same as I would with the other module. I checked with Shay in the Plum Audio group - he replicated the behaviour and thinks it is a firmware bug

patrickdowling, 2020-04-19:
Just to clarify: It works on the normal o_C, but not the OCP?

djrrunstein, 2020-04-19:
Hi Patrick,

Thanks for replying. Yes, I've had the other o_C for over a year or more, and it works just fine on that (it's running 1.3.4) The OCP is running 1.3.6 and it doesn't work

David

patrickdowling, 2020-04-19:
Ok. It's possible there's something that needs adapting to the VOR that wasn't included in the original [pull request](https://github.com/mxmxmx/O_C/pull/90) from @Chysn, but honestly I've not been involved in either.

djrbrunstein, 2020-04-19:
Thanks Patrick. So, as far as you're aware the autotune function in 1.3.6 works as it should on normal o_Cs? I know next to nothing about coding, and have never used GitHub before today, so don't really know the best way to proceed with this. Thanks for your help so far.

patrickdowling, 2020-04-19:
Yeah, from a quick scan it doesn't look like there were any changes to the autotune recently, and I don't recall anyone complaining. So my assumption is that it should work but I don't have proof either 😉
Seems like there's two approaches: Verify that it does indeed still work on regular ol' o_C with 1.3.6, and check if the feature works on any VOR platforms.

djrbrunstein, 2020-04-19:
Hi Patrick,
I've updated the normal o_C to 1.3.6 and all works as it should. Do you think that means that this is a question for Shay?

patrickdowling, 2020-04-20:
Awesome, thanks for testing that. My suggestions would be that Shay test if it works on other VOR hardware if he hasn't done so already; it's probably a firmware issue but if it's specific to those variants then arguably it's his responsibility to get fixing it onto someone's docket.

djrbrunstein, 2020-04-20:
Nice one, I'll let him know and see if he's able to replicate this on 4 Robots or something. I might see if anyone else in the Facebook group can test too.

patrickdowling, 2020-04-21:
The perhaps super-obvious thing would be that it depends on the output range; I'm not totally clear on how that should work is anyway.

djrbrunstein, 2020-04-22:
Yeah, I tried all the various ranges but with no difference. I've mentioned it to Shay now, so hopefully this will be addressed before too long