mxmxmx/O_C #102: My O&C won't start after trying to update firmware

baunder, 2020-07-17:
Hey people, I just tried to upload the Hemisphere firmware into my O&C and I got an error, and now the module won't even start. I tried looking everywhere for troubleshooting, but couldn't find much.
I followed every step in how to update firmware, but I got an upload error on the teensy app (OSX Mojave) and now it won't even let me reboot it. It's dead.
I've got no experience nor equipment for testing the module, so is there anyway I can reboot it?  
Maybe restore it to a previous version? 

Sorry for all the questions, but I'm a newbie, but have done firmware upgrades in the past. I just don't know what went wrong.

Any help is appreciated!
Thanks
Paul

patrickdowling, 2020-07-18:
There's no real fallback other than the built-in bootloader on the Teensy. That can't normally be overwritten so as long as the Teensy isn't defective the loader application should be able to communicate with it. The closest to restoring would be to just upload the regular .hex file from the [releases page](https://github.com/mxmxmx/O_C/releases). 

How is the module powered? If the Teensy jumper was cut, you'll need to power it externally. If it wasn't cut, don't power it via USB and modular power at the same time (see [build it](https://ornament-and-cri.me/build-it/) or ask whoever built yours).

IIRC there was some "boots to black screen" issue on Hemispheres but you'd have to check [their page](https://github.com/Chysn/O_C-HemisphereSuite/) for that.

baunder, 2020-07-18:
Thanks Patrick! It ended up working after 7 tries, but now the modules turns off randomly. I'm going to restore to the original firmware to see if it's solved. 
Thanks for your time

MichiganSynthWorks, 2020-07-18:
I can confirm this behavior on some hemisphere builds, it looks like its going to sleep but it wont wake up.  Had one here for debug and I couldnt replicated it on my setup and everything checked out with the module, it was still going it on the users rig after i returned it.  Removing Hemisphere back to stock 1.3.6 resolved the issue so it looks like there is a bug there somewhere and that should be taken up with those guys

baunder, 2020-07-19:
@MichiganSynthWorks I will try returning to 1.3.6 and see what happens. Thanks a lot!