mxmxmx/O_C #104: Optimization question

dylanreimer, 2020-07-30:
Hello, 

I was just wondering what effects smallest code optimization has on the behavior of the firmware. I'd like to flash with serial usb type so I can flash in the future without having to manually enter program mode by pressing the program button, but the sketch currently is ~500 bytes too large. 

Thanks,
Dylan

patrickdowling, 2020-07-31:
The somewhat generic answer is "it depends". The -Os setting changes the runtime behaviour and that's already borderline in some cases (see issue #98) so it's a bit of a gamble. IIRC there was a reason why it's not used but it's lost in the fog...

That said, if you use the [fix/display_glitch](https://github.com/patrickdowling/O_C/commits/fix/display_glitch) branch, that not only fixes the glitch, but I added some minor tweaks that save some flash space.

With some major refactoring I got things down closer to 80% usage but that has serious side effects.

dylanreimer, 2020-07-31:
Wonderful, tested with firmware from the new branch and it's working and flashing great. Thanks so much for the help!

closed 2020-07-31 by dylanreimer