mxmxmx/O_C #106: Unable to long press during CV Input Scaling calibration

ChrisWeiss, 2020-12-03:
Using 1.3.5 downloaded from this releases page, I seem to be unable to use long-press to adjust the CV Input scales during calibration. All of the other calibration elements work. This appears to be needed as the display is reading ~400 for the 1v input scale. Holding down the right button does nothing, releasing it moves to the next scale for calibration.

Anyone else seeing this? Should I just re-flash?

patrickdowling, 2020-12-03:
IIRC the long press there just stores the current ADC reading and moves to the next calibration step, so what you're describing sounds normal. ~400 also sounds about right -- it's not supposed to be 0, but (again IIRC) ~4096/10 x 1 -- so unless there are tracking issues the 1V/3V calibration is probably optional.

ChrisWeiss, 2020-12-04:
Ahhh, my bad, I was used to the "long press to adjust" workflow and didn't grok "and long-press the right encoder to *record* the offset value". Thanks!

closed 2020-12-04 by ChrisWeiss