mxmxmx/O_C #107: Feature request: attenuator and offset as an extra parameter to apps doing sampling

giacecco, 2020-12-06:
Hi All, I am a newbie so just destroy my message if I am saying bulls*it 😀 

Many O&C's apps include a sample & hold element of the input CV, to which one would typically feed noise to. But the range of noise source - I typically use Mother-32's +/-10V - is in most cases much wider than I need, and I need to attenuate it. 

Utility modules such as Expert Sleepers' Disting cater for that problem by offering attenuation and offsetting of the CV input. I was surprised to see that the O&C's apps such as Quantermain do not do the same, also because it feels intuitively easy to implement. It must be a design choice, but why?

Also, it feels like a waste to have to attenuate and offset noise before going into the O&C, that is already doing incredible things of its own... but the simplest one! I feel like I'm missing something.

Thanks!

G.

patrickdowling, 2020-12-06:
Well the simple answer is that no one felt inspired to implement it (except, I believe, in _CopierMaschine_).

It's a design decision insofar as the menu lists are all fairly long without adding something that can be done easily outside the module (arguably better with knobs). Given the feature-creepy nature of how things evolved, "easy to implement" isn't necessarily the case any more -- especially if it's to be unified across apps -- and we already had to start sacrificing scales to fit the code (*).

That said, I did a (hereto unreleased) major framework refactoring that included input attenuvertion in a dedicated IO menu, but it's a major time investment to adapt and especially test all the apps.

(*) There's a bit of space now that the issue with newer gcc versions has been identified.

giacecco, 2020-12-06:
I understand the feature creep issue, and the refactoring you described seems like it will be a very welcomed addition. Thanks!

closed 2020-12-06 by giacecco