mxmxmx/O_C #111: Quad quantizer transpose with aux CV

bobbijvoet, 2021-05-16:
First of all, thanks for the amazing work on O_C. Love it so much!
I have a question about the quantermain app.

I'm using a 1V/oct source for transposing melodies on multiple modules (like the arpitecht). 
Now I'm trying to use that same voltage to transpose a channel in quantermain using CV aux, but it seems like it is not using 1V/oct. 
Is this by design? Can I somehow make it follow 1V/oct scaling?

patrickdowling, 2021-05-16:
No, the aux transpose doesn't use 1V/oct. Since it's not a chromatic transpose but in scale degrees, i.e. it always within the selected scale/mask, that's kind of by design (none of the aux/modulations are really calibrated to anything in particular).

bobbijvoet, 2021-05-16:
Thanks for the quick answer! 
I see and it indeed makes sense that it's always within the selected scale/mask. 
Still it would be nice if it was calibrated. 

I would like to try to change the source and compile it so I would have 1V/oct control.
My guess is that I would have to do some additional calculation on the incoming analog value: https://github.com/mxmxmx/O_C/blob/659f97e09145c526c89839a30714782e99ab8fc7/software/o_c_REV/APP_QQ.ino#L741.
Is that correct? Any other pointers maybe?

patrickdowling, 2021-05-16:
I guess it depends what exactly you want to control. You can maybe just run the value through an instance of [SemitoneQuantizer](https://github.com/mxmxmx/O_C/blob/master/software/o_c_REV/OC_scales.h#L35) with something like `quantizer.Process(OC::ADC::raw_pitch_value(channel))` but I don't think that's going to be any more intuitive. That value is just used as an offset in the lookup table so doesn't map to semitones, and unless I'm missing something it's already fairly close (10V/128 = 0.08V per step, vs. 0.083V for 1V/oct).

More common approaches seem to be adding either some number of semitones to the pitch value before or after quantizing (the latter being a chromatic transpose, the former you don't always get a different note but it stays in scale/mask).