mxmxmx/O_C #112: Request - CPL or Pick and Place file for the PCB

mlinton, 2021-05-23:
Is there any chance that we can get the CPL or pick and place file for the SMD components for assembly at a fabrication facility like JCLPCB?  Alternatively posting the Eagle or Kicad files would allow us to generate this for easy export as well.

Thanks again!

patrickdowling, 2021-05-23:
I'm fairly certain the 14HP PCBs were layouted with Fritzing. Not sure if that's going to be of much use in that regard.

timchurches, 2021-05-23:
Please see the licensing informations at https://ornament-and-cri.me/licensing/ - if you are planning automated production, you need to discuss this with Max Stadler first, he holds the copyright to the hardware design. I can put you in touch with Max if required (I know, his email address is not actually listed on the O&C web site).

frequenzteiler, 2021-05-24:
sorry , if im not asked but contribute to this topic . who will have access to those files at the manufacturing site ? this does not feel right . better try to keep mass production out of CC licensed and diy oriented projects ?  of course thats just my very personal opinion.