mxmxmx/O_C #113: How to build on 64-bit machines

benirose, 2021-11-10:
I'm following the build instructions listed [here](https://ornament-and-cri.me/firmware/) under "method B", but it specifies requiring v1.35 of Teensyduino. The issue is that this seems to be a 32-bit application (or at least the installer is 32 bit) and I am unable to install it on my mac running Catalina. I understand the issue is the version of gcc used by the IDE. I am currently running Teensyduino 1.53 (built on Arduino IDE 1.8.13 it seems) and I'm wondering if there's a way to swap out the gcc version used by the IDE or any other alternatives to finding an older box to build this on. Are there any plans to upgrade the code to support building in > gcc 4.x?

patrickdowling, 2021-11-10:
The code should _compile_ but while newer versions generate a binary that is smaller, it's apparently also a tad slower and triggers what was a rare underlying bug almost immediately after start. It's a bit hidden in https://github.com/mxmxmx/O_C/issues/99 but AFAIK [this branch](https://github.com/patrickdowling/O_C/tree/fix/display_glitch) works just fine with newer Arduino/teensyduino versions.

benirose, 2021-11-10:
Thanks! I was able to use my work laptop running Mojave to compile my branch. I have yet to have any issues compiling on Catalina, but when it uploads to the Teensy I just get a blank screen. I actually got the same thing the first time I uploaded from Mojave, but cycling power fixed it. It did not seem to fix it from the Catalina upload. I'll take a look at that issue and branch, thanks a ton!

closed 2021-11-10 by benirose