mxmxmx/O_C #114: Change the name

oliviaartz, 2022-04-13:
Stop associating this useful open-source tool with that racist essay.

timchurches, 2022-04-14:
Have you read any Marx? Or any of Joseph Conrad's novels? Or Dickens? Or just about any European thinker from the 19th and early 20th Centuries? We are not defending the racism that pervaded so much of the thought and writing of those times, but nor do we believe that ideas that may have been expounded in writings that also reflected the racist prejudices of their time should be expunged from history. 

O&C is open-source, and you are free to fork the project and rename it if you wish.

oliviaartz, 2022-04-14:
…as fascinating as that reply is I still think you should dissociate this great project from that bumbling, racist tirade by a dead sexual predator.

timchurches, 2022-04-14:
Sure, I understand your position, and am not dismissing it. Max (aka mxmxmx), who suggested the name, and who is an historian of science and technology, would have something to say about this (and he would certainly not dismiss your point), but he is on an extended break right now and not readily contactable. The name was certainly intended to be ironic. We'll discuss it when he is back online.

patrickdowling, 2022-04-14:
Yeah, I don't end up seeing any need to rename but we can review it. One might argue the module pokes fun at the source -- 
is it even mentioned explicitly any more? -- or has even reclaimed the phrase with positive spin (at least within this niche). Which gets close to "Why should we have to change, he’s the one who sucks” territory.

mxmxmx, 2022-04-14:
... surprised it took 8 years or so for someone to bring it up — anyways, point taken, that wasn't a particularly wise choice naming it like that; I'm just not sure to dissociate it now, other than deleting any reference to Loos and/or said essay here on this site. Since it can't be undone, I think I'd prefer being explicit about it; there's any number of instructive takes on "Ornament und Verbrechen", I'm happy to turn the "Readme" into a reading list, so people can read up should they be so inclined. Or what would you suggest?
 
( and hi Tim / Patrick, ... sorry I've been very out of the eurorack loop )

patrickdowling, 2022-04-14:
> Since it can't be undone, I think I'd prefer being explicit about it; there's any number of instructive takes on "Ornament und Verbrechen", I'm happy to turn the "Readme" into a reading list, so people can read up should they be so inclined.

FWIW I think that's a good plan.

(hey! It has been a while :))

oliviaartz, 2022-04-16:
No need to undo anything. You can stop associating this cool project with that bafflingly terrible essay. By changing the name and updating the code, PCBs, etc. At this point else is just kind of bending over backwards to keep the association in tact.

oliviaartz, 2022-04-16:
Sounds good to me too. Looking at O_C today it looks sparse and almost minimal. Like you were following his advice! 8 years ago a display was the height of ornamentation so you were expressly giving Loos the finger. If I understand what I’m hearing. If this is the case, that’s probably worthy of a not in 2022.

fwiw, im glad to have gotten the band back together even if it was in reaction to me saying “hey! wtf?!”

oliviaartz, 2022-04-29:
Might make sense to rename this issue to “Address relationship to essay” or something. But I think I’ll step back and let you folks take it from here.

skinlayers, 2022-05-27:
Side note: I ironically named my goth/industrial DJ night after this module, because a) O&C was influential on the Bauhaus school (which is also the name of one of the first goth bands), and b) because most goths are all about ornamentation, and thus pretty much the antithesis of the Loos' rant. ;)